MobiFlight/MobiFlight-FirmwareSource #36: Remove duplicate copy of config buffer

danecreekphotography, 2021-09-20:
From https://github.com/danecreekphotography/MobiFlight-FirmwareSource/pull/13#issuecomment-921924372.

Currently a copy of the config buffer is made because strtok_r replaces the delimiter in the original string with a null character. @elral has a proposed solution that will remove this duplicate buffer copy.

elral, 2021-09-20:
The idea is only to keep the config once, process this config (replacing the delimiter with NULL) so that all pointers from the devices get there information and do not keep a copy with the original config. This duplicate config is "only" needed if the connectors requires the config vai `void OnGetConfig()`. Instead of sending the duplicate buffer this function now sends directly the content from the EEPROM byte wise. But this would mean also that the CMDMessenger has to be extended by a function where bytes can be send w/o field separator. In my fork I have this already added, and the changes required for MF I will propose soon.

elral, 2021-09-21:
I have to branches with slightly different changes. Both have advantages and disadvantages to discuss. Should I for both make a pull request?
And it would be really good if some one could also test these changes.

danecreekphotography, 2021-09-21:
Open two pull requests, the build system will automatically produce the firmware and attach it as a comment to the PR so it's easy to download and test.

commit 41bccb27 (2021-11-07): Remove duplicate copy of config buffer
Fixes #36
commit d9e8d66c (2021-11-07): Remove duplicate copy of config buffer
Fixes #36