MobiFlight/MobiFlight-FirmwareSource #100: Support Arduino Nano

neilenns, 2021-11-21:
Firmware companion to https://github.com/MobiFlight/MobiFlight-Connector/issues/579.

This will require:

* A `board.h` file based off the Uno board
* A new entry in `platformio.ini` to generate Nano builds

No change to the GitHub build process is required, it was designed from the start to expect more Arduino boards in the future.

elral, 2022-03-07:
Last days I got an Arduino Nano clone, and I figured out that this clone has still the old bootloader installed in contrast to original Nano's which have a new bootloader.
Within Platformio.ini one can differ between both, but I am afraid that users will run in similiar problems like with the Uno clones unless there is solution how to explicitly choose a board in the Connector for flashing the firmware.

tigert, 2022-03-17:
I flashed my old bootloader nano with commandline avrdude, and comparing to what Arduino IDE does versus what MopbiFlight Connector does, the difference seems to be this: -b57600 and -carduino - these work for my nano clonses.

To recap a bit of discussions on Discord arouind this topic, the most pressing issue is those nanos use the same VID/PID as most megas (its likely the CH340 serial chip that most chinese clone arduinos have that has the same ids on all of them. So we need to have some heuristics based on what you could try as the most common board first (mega?), see if it fails, and try the next one (nano/uno?) etc.. As long as it can be sorted out, once the board is flashed it will work.

elral, 2022-03-17:
The old nano bootloader requires -b57600 where the new nano bootloader requires -b115200. All other parameters are the same as @tigert mentioned.

It seems indeed that some "testing" which board is present might be required.

PR #223 (2022-12-08): Add Nano support
## Description of changes

Fixes #100 

* Add nano to `platformio.ini`
* Add `MFBoards.h` for the nano device

The board file came from @elral.
closed 2022-12-08 by DocMoebiuz