MobiFlight/MobiFlight-FirmwareSource #103: Improve performance of writing to LCD Display

MobiFlight-Admin, 2021-11-25:
Optimization of LCD Screen Library still required.

elral, 2021-11-26:
I already made some changes to speed up the library, it's now in the range of the HD44780 lib. But some more tests are required, and maybe some more optimazations.
It seems that it is not going much faster than 20ms for 80 characters, so it might be that more measures than optimization the LCD lib might be required.

MobiFlight-Admin, 2021-11-26:
@elral that sounds amazing! i have changed MF so that we can send a string together with a starting position (line, col). This would allow to send only two modified parts of the text. The logic on the Connector side to break it up in pieces is still missing. the Apply function assembles the text respecting all placeholders and such but it is called before the "display" method is called. I didn't find a good way yet to change that. Need to think about it. I also don't think that it would make sense to simply send the placeholders down. There are situations like: 
Squawk # # # # - basically a space between each digit.

we could also compare the current text with the text to send and detect the diff.

ideas are welcome!

MobiFlight-Admin, 2021-12-04:
Successfully closed, see https://github.com/MobiFlight/MobiFlight-Connector/issues/346 for more information

closed 2021-12-04 by MobiFlight-Admin