MobiFlight/MobiFlight-FirmwareSource #117: No pin check upon definition of OutputShifters

GioCC, 2021-12-08:
When Output shift registers are defined, no check is performed in the firmware on whether specified pins are already assigned to other blocks, nor they are marked as used by the shift register (as is done for all other block types).

**It should be considered whether these checks - and the whole recording mechanism for pin allocation - still make sense at all (in the firmware)**, given that they occupy a non-negligible amount of memory.
Of course consistency checks are sane and very good practice, but since the only interface for the configuration of the firmware is the PC "connector", checks could be entrusted to it (they have to be performed for the benefit of the user interface anyway) avoiding redundance.

neilenns, 2021-12-08:
Personally I would *love* to get rid of this check in the firmware.

neilenns, 2022-01-06:
@GioCC I think I'll take a few minutes today to try removing this stuff and see how much ram/flash it frees up.

neilenns, 2022-01-06:
[image]

GioCC, 2022-01-07:
Granted, not a huge impact (still welcome though, why waste?); however it's also very much about consistency.

neilenns, 2022-01-07:
@GioCC I agree it's not huge but we are so constrained on RAM and flash that every little bit helps. This check is already done on the desktop side, it seems unnecessary to spend the RAM and flash doing it as well on the firmware side.

GioCC, 2022-01-08:
@neilenns Yes, exactly my first point. And I think this topic deserves a good deal of focus in the next changes.

elral, 2022-02-07:
The pin check mechanismus is eliminated with merging the pull request #147 . So this issue can be closed too.
Unfortenutely I forgot to mention this issue in the pull request.

closed 2022-02-25 by GioCC