MobiFlight/MobiFlight-FirmwareSource #136: Get rid of MFOutput.cpp?

neilenns, 2022-01-08:
In looking at the code for unrelated reasons I realized that the `MFOutput` class is pretty much useless. The only place any of its methods ever get called is for power saving. Seems like we're spending an awful lot of RAM and Flash overhead to maintain a class that only gets used for one method.

Perhaps we can do this in a more streamlined way with basic bit arrays? One to indicate whether a pin is in use as an output and the other to save the last set value of the pin for power save restore?

neilenns, 2022-01-08:
@GioCC have you ever explored this?

neilenns, 2022-01-08:
My first quick and dirty attempt using a library that implements a `BoolArray` was not promising at all. Saved a teeny bit of ram at the expense of quite a bit of flash.

elral, 2022-01-08:
Hmhm, you have to store the pin number anyway. Additionaly the old status is stored, which could be also a PWM value. So a bit array seems indeed not to have any advantage for the outputs.

I see a bigger chance to save RAM for the MFButton.cpp. As far as I remenber for the actual status and the old status one byte is used, whereby only 1 bit would be required for each status. In theory you could save 1/4 for buttons.

The same should applies for the encoders, even that the sig1/2 is definded as bool. Each bool needs also 1 Byte!

GioCC, 2022-01-08:
@neilenns , no, i haven't yet.

About the use of a "BoolArray", I used a similar class several times in other projects; it was also in the MF overhaul I proposed to Doc long time ago. 

Theoretically, I'm confident it might still be beneficial, but there are several drawbacks:
1. it severely increases coupling between objects
2. in order for it to be fully exploited, the impact on the implementation of existing objects is not negligible

In the case of current Mobiflight, I think these drawbacks question the potential gain, although I'm not sure they are enough to completely rule it out; it just calls for a more comprehensive evaluation (and possibly not is a "fluid" development situation as the current one).

One idea could be to use a common"bitStore", as I used to call it, both for buttons (inputs) and outputs; the access index for both could be the pin number, which is already stored. This could spare one net byte per In/Out (actually, 7- bits), if it weren't for the fact - as @elral correctly points out - that outputs must manage PWM values. (then of course there's the flash space overhead)

closed 2022-01-09 by neilenns