MobiFlight/MobiFlight-FirmwareSource #159: Introduce an official setting for the VS Code C/C++ autoformatter

GioCC, 2022-02-24:
In order to have a consistent code formatting, avoiding the recurring discrepancies in commits, we could conveniently use the automatic formatter of the standard "C/C++" plugin of VSCode. 
This could be accomplished by having a `settings.json` file saved in the `.vscode` directory; attached is a sample with my current settings.
Reformatting can be made effortlessly by explicit command or automatically upon file save.
I'm not really sure about this last option; it might confuse the user, but it would be a useful safeguard.

[settings.json.txt](https://github.com/MobiFlight/MobiFlight-FirmwareSource/files/8132191/settings.json.txt)

GioCC, 2022-02-24:
Link to [CLang formatting options](https://clang.llvm.org/docs/ClangFormatStyleOptions.html)

elral, 2022-02-25:
For me it is important that we all have the same settings, however they are. I am not very familiar with these settings, so I will use what every body else thinks is sensefull.

elral, 2022-03-09:
I tried this on my firmware for the motorized Trimwheel/Throttle and was wondering, that nothing has changed. Now I found with Autosave enabled you have still to actively save the file for reformatting.

I tested the settings and I am fine with them, would be good to have them in the near future for the next changes ;)

GioCC, 2022-03-09:
> /edit: It seems not to work with .h files, is this intended?
Hmmm, never noticed that... not 100% sure, but I'd swear it works for me... I'll check.
 
> /edit2: Oh, and these settings change the order of the `#include` files ...
Yes, that's something I'd like to rectify too. There's obviously a rule to it, but changing the order of includes might break compilation at times. Will look for that too..

elral, 2022-10-06:
Do we have the common understanding that the merged file with the autoformatter settings are agreed?
What about the .h files, seems still not to work for me after a short test.

DocMoebiuz, 2022-10-09:
Important to agree on this @GioCC and @elral.

GioCC, 2022-10-26:
@DocMoebiuz @elral @neilenns Guys, do you think we can close this issue considering the file already in the repo as accepted? Are there any other questions/issues?

elral, 2022-10-26:
For me it is fine, I checked also that the .h files are working for me.
However, this will us not protect against having changes on not changed code parts. If we add e.g. `#define EXAMPLE_TEST 200` to a `#define EXISTING 66`, then `#define EXISTING` gets filled up with spaces.
This will happen also on other parts.

It was merged with PR #167

GioCC, 2022-10-26:
@elral exactly, good catch. The PR was about reformatting, so I guess I thought at the time that it would be advisable to have a formally defined setup, exploiting the capabilities of VSCode; regarding its rules, it has been pending explicit approval since then.
I took that merging the PR was a sort of approval, although of course I'd be happier if this were finally and clearly agreed.

Regarding files that are still unaffected, a full reformat pass and a new PR could solve that for good.

simontrhodes, 2023-09-20:
I'm really interested in this because there are any published coding standards.

elral, 2023-11-08:
The settings for the autoformatter was merged with PR #167.
So this issue can be closed.

closed 2023-11-08 by elral