MobiFlight/MobiFlight-FirmwareSource #165: Demo message can prevent programming of older Megas

GioCC, 2022-03-08:
I had a strange issue lately: a firmware that compiled flawlessly could not be flashed on one (or several) of my Megas.

The flashing process started regularly but got stuck every time towards the end:

```
Looking for upload port...  
Auto-detected: COM12  
Uploading .pio\build\mega\mobiflight_mega_0_0_1.hex  

avrdude: AVR device initialized and ready to accept instructions  

Reading | ################################################## | 100% 0.01s  

avrdude: Device signature = 0x1e9801 (probably m2560) 
avrdude: reading input file ".pio\build\mega\mobiflight_mega_0_0_1.hex" 
avrdude: writing flash (27012 bytes):  

Writing | #################################################avrdude: stk500v2_ReceiveMessage(): timeout 
avrdude: stk500v2_ReceiveMessage(): timeout 
avrdude: stk500v2_ReceiveMessage(): timeout 
...
```
After some digging on Google, it turned out that some older Mega firmware versions would detect a sequence of three exclamation marks in the data stream and interpret them as a switch to a terminal mode, thus interrupting the upload process (details [here](https://www.switchdoc.com/2013/12/bootloader-issues-with-arduino-mega-2560/)).
Searching the code, there is indeed a similar sequence (`0x21 0x21 0x21`) towards the end of the hex file, which correspond to the `"Mobiflight rocks!!!"` string in the LCD class.
Removing at least one of the exclamation marks solved the problem.

The link above also suggests that in more recent (or different) versions of the bootloader this problem is solved.

Proposed solutions:
- curb the enthusiasm when declaring Mobiflight qualities to LCD users;
- upgrade the Mega bootloader

The first solution is probably worth following anyway.

Some more links:
[Arduino forum](https://forum.arduino.cc/t/bug-with-using-mega-2560/47107/9)
[Google Code](https://code.google.com/archive/p/arduino/issues/392)
[AVR developers](http://www.avr-developers.com/bootloaderdocs/index.html)

neilenns, 2022-03-08:
This is an amazing bug, way to chase down what caused it!

elral, 2022-03-09:
Yes, I read about this some weeks ago while searching for a complete other topic. But if you had asked me what the problem might be, I hadn't recall this.

DocMoebiuz, 2022-03-09:
oh wow. thanks.

commit 134ce394 (2022-03-09): Demo message can prevent programming of older Megas (#166)

Fixes #165