MobiFlight/MobiFlight-FirmwareSource #174: Removal of the fixed per-device limit of configurable devices (and associated data structures)

GioCC, 2022-04-23:
This issue refers to discussion #154.
Basically, the "index" arrays for each device type are suppressed, and devices are allocated completely in the already existing object  buffer.
This way, there is no longer a limit to the number of devices of a given type that can be allocated (e.g.: one can allocate exclusively shift registers, as many as the avaliable space allows). The max amounts need no longer to be defined.

DocMoebiuz, 2022-10-09:
@elral can you please also comment on this? I think it would be great to have this.

Open questions:
* is actually no limitation for devices always sensible? Do we not also depend on refresh speed? I think it would be OK to not worry about this and leave it up to the user to decide what is acceptable.
* Does the UI need any modifications? Right now it knows that it cannot add an additional device. This would have to be modified because we don't upload the config immedieately every time we add a device. So the UI would have to keep track of the used RAM too.

elral, 2022-10-10:
This would give complete flexibility to the user which and how much! devices he want to use on a board.
As every device is affected this will be a lot of changes and we should do tests before merging.
But I am also thinking about if processing time could be affected if e.g. a user defines on a Mega "a couple" of inputshifter/multiplexer. Do we want to allow a mega with input shifters on all pins? Or more than 2 LCD's as they need much more time than we need for reading the encoders. So it might be possible that encoders are not read proberbly if lot's of LCD are connected (limited by the available I2C addresses).

I will not say that it should not be done, but I think that still a limitation of max. devices is required.

On the connector side I these the modifications as Sebastian has already mentioned. For now the firmware reports a `kStatus` message that the buffer is full, but I think this is not evaluated for now in the connector. Maybe this should be changed in a new `kError` message or at least processing the existing buffer full message.

GioCC, 2022-10-10:
I fully second all of @elral 's comments. Even though we of course don't want to allow a mega with input shifters on all pins, and however unlikely we assume that to be, this would be definitely a possibility unless restrained.
"Heavy" peripherals like LCDs could probably be limited to a reasonable number anyway (whether that means keeping the current storage implementation or handling them the new way), but more "common" peripherals may add up to unwanted scan times.
I also find @DocMoebiuz 's proposal basically correct ("leave it up to the user to decide what is acceptable"), maybe with some safeguard limit added.

About connector modifications: right now, I'm not sure whether there are cases where, even with assigned max capacity, a configuration exceeding the firmware capacity could be entered (and failing upon upload); after all, current capacities are designed to accomodate a sufficient number of instances for each device, but I'm not sure whether it can be fully exploited.
But regardless, it would be very good if a way could be found to avoid having a redundant capacity calculation in the connector.

Regarding encoders: I think this might deserve a separate topic (yet to be fully explored) about prioritizing those digital inputs where encoders are attached with more frequent scans.
This could even make feasible attaching encoders to shifters/muxes, but their connections should not be too sparse (maybe requiring them to be attached to adjacent inputs?).

GioCC, 2022-10-10:
And another point I'd like to stress once more (even though obvious): this modification is not quite trivial, therefore it would require an additional testing effort. Nothing groundbreaking, but a little more attention and trial time are in order.

Besides, I closed the initial (draft) PR a few days ago because it was going very out of sync with the main branch; if there is further interest in evaluating it, I will bring it up to date again and make it available.

elral, 2022-10-10:
Regarding the encoders there cold be a solution, but I guess Sebastian is not really happy with this ;)
I proposed it some month ago to read the encoders within an ISR, so they get read every 1ms unrelated to any device which might take longer in execution. But this would mean up to split the encoder routine into just reading the inputs and evaluating the changes within a separate function which is called from the main loop (that's similiar to other input devices, `poll()`).
I tested this a couple of month ago and didn't see any effects especially to the serial communication. I will redo this with all the merged pull requests from the past.

GioCC, 2022-10-10:
That would be an interesting solution, definitely it would be a clear advantage with regard to scan time consistency.
However I'm still bothered with the issue of sparse connection: as long as encoders are limited to direct pins (which could be an option) I think it would be good, but if we are interested in allowing encoders to be connected to Shifters/Muxes (are we?), things would become quickly more difficult.

elral, 2022-10-10:
Maybe we should first try to evaluate what the max. numbers of devices could be. And in a next step to "calculate" the required memory consumption if we stay with the existing arrays while increasing the size to the max. devices. Then we can compare it to the memory required which this change need (I think there will be some, didn't look into GioCC code for a longer time) and decide, if the effort is worth.

> I think it would be good, but if we are interested in allowing encoders to be connected to Shifters/Muxes (are we?), things would become quickly more difficult.
I think also it would be good to have encoders connected at least to the inputshifter (I am not so familiar with the muxer). But this does not argue against an ISR, even the shifter could be read within the ISR. If it takes too much time for the ISR, it will be also a problem if reading within the loop() as for now.
Connecting encoders gets much more interesting for the Raspberry Pico with limited I/O's compared to the Mega (in my eyes). But this one is running much faster so it should be a less problem.

GioCC, 2022-10-10:
> [...] "calculate" the required memory consumption if we stay with the existing arrays while increasing the size to the max. devices. Then we can compare it to the memory required which this change need
I don't know if this comparison is really useful. There is _very_ little additional RAM usage (almost negligible IIRC), but the same amount of memory would allow even a large difference in the number of devices. For a first estimate, assume no additional RAM required (or, like, two bytes for each device, to be on the safe side), and imagine e.g. removing the RAM of LCD, steppers and servos and filling the gained space with LED displays and shifters.

> I think also it would be good to have encoders connected at least to the inputshifter (I am not so familiar with the muxer).
To all out practical intents and purposes, you can consider the muxes as shifters.

> But this does not argue against an ISR, even the shifter could be read within the ISR. If it takes too much time for the ISR, it will be also a problem if reading within the loop() as for now. 
Reading shifters/muxes in an ISR would block the serial IRQs, with worse results (missing chars) than doing it in the main loop (safe as long as buffer does not overflow).

PR #253 (2023-11-04): Get rid of fixed device numbers
## Description of changes

The number of devices of each type will be calculated before the config gets loaded from the EEPROM.
With this information the device buffer gets filled up with the defined devices.
The device index array isn't needed anymore, instead a pointer to the first device of a type is stored.

fixes #174 

Notes:
* How to handle a full device buffer. For now it can happen that the device buffer gets filled completly, but during loading the devices this is not considered.
* As the device index arrays are not needed anymore, this should be a hug [... cut]
closed 2023-11-04 by DocMoebiuz