MobiFlight/MobiFlight-FirmwareSource #180: Increase number of encoders for Uno and ProMicro

elral, 2022-05-08:
As proposed in https://github.com/MobiFlight/MobiFlight-Connector/issues/784 the max. number of encoders could be increased to 9 to use all available pins from Uno ans ProMicro. This will require additional 8 Bytes which is OK due to all memory optimazations which were implemented in the past.

elral, 2022-05-08:
While checking the max. devices for the Uno and ProMicro, they still differ for some devices.

Uno:
#define MAX_STEPPERS        2
#define MAX_MFSERVOS        2
#define MAX_DIGIN_MUX       2

ProMicro:
#define MAX_STEPPERS        3
#define MAX_MFSERVOS        3
#define MAX_DIGIN_MUX       1

@MobiFlight-Admin @neilenns @GioCC should we harmonize it or leave it as it is? The required RAM consumption for the Uno is negligible (4 Bytes more for increasing stepper and servo).

elral, 2022-05-08:
Hmhm, for the Uno I have to do some math if max. devices will fit into the device buffer before changing!

elral, 2022-05-09:
Increasing the max. numbers of encoders to 9 should still fit into the device buffer. Same for the steppers and servos.

BUT, I tested it for the ProMicro and some strange things happen.
After configuring "some" encoders (the number of encoders are not always the same) the encoder do not work anymore until the ProMicro is hard resettet. A configured button is still working and the config can still be read without hard reset.. So the ProMicro does not crash. For now I have no idea what's happening. Some more investigations are required to find out what's going on.

GioCC, 2022-05-09:
Hmmm... interesting. Is this specific for the ProMicro, though? (I imagine this should hold for a Uno too).

elral, 2022-05-09:
Can't answer this in the moment, I have to solder an encoder first to my Nano (don't own an Uno). Also an Uno makes an hard reset on reconnect, so difficult to find out. I have to use a terminal and do test it "by hand".

elral, 2022-05-09:
The device buffer gets not cleared if the config gets resetted. So after changing the config and uploading the new config some/all devices will not be loaded as the device buffer might not have enough space.

Without this bug 9 encoders are working properly.

PR #183 (2022-05-26): Increase number of encoders for Uno and ProMicro
The max. number of encoders is increased to 9 for Uno and ProMicro. All available 18 Pins can be used for encoders.

Fixes #180
commit 68e8bd8c (2022-05-26): Increase number of encoders for Uno and ProMicro (#183)

Fixes #180
closed 2022-05-26 by DocMoebiuz