MobiFlight/MobiFlight-FirmwareSource #200: Mega get's sometimes not properly identified

elral, 2022-10-03:
Sometimes the Mega (and I expect the same for the Uno) get's not properly identified from the connector. That means that:
* eeprom config
* serial number
* devices
can be misread (not always). This was found from @Jaime Leon and is reported on Discord.
He found also, that all switches put into OFF position helped for properly booting.

It seems that this behaviour is related to the initilisation of the buttons. Every button is initilised woith `1`, so after startingup for each ON button a status change is detected and a serial message is generated. For multiple buttons multiple messages are generated. As this comes directly after the GUI has connected (reset via DTR for Mega/Uno) and the connector asks for the board info's, it seems that this disturbs the correct identification of the aboved mentioned infos.
A first test from Jaime Leon of initialisation the buttons with the actual status solved this problem. As no status change is identified on startup no serial message gets generated.

The same applies to:
* input shifter
* multiplexer
* analog in
 For analog in this is fixed by PR #199 where the average buffer is filled on startup. For the input shifter and buttons I have already prepared the changes.
For the multiplexer I would like to ask @GioCC to prepare the changes as I am unsure what the best way would be.

GioCC, 2022-10-03:
OK, I will have a look as soon as I'm able. Probably the multiplexers require the exact same changes as the input shifters, so it would be useful to see how you implemented it in that case, in order to keep the two consistent.

GioCC, 2022-10-03:
Not really sure if duplicating the shift register poll in the constructor is nice; also it's not consistent with the approach used for analog inputs (use the existing function by adding an argument  that says whether the change of value should trigger events).

GioCC, 2022-10-03:
How to proceed, should I open a different issue to to add the multiplexer fix, or should I reopen this?

elral, 2022-10-03:
I am fine with both with a tendency to reopen this issue.

elral, 2022-10-04:
> [(](https://github.com/GioCC/MobiFlight-FirmwareSource/commit/c7a513768490a0f39ec852982837daff50c02284)https://github.com/MobiFlight/MobiFlight-FirmwareSource/issues/200 [follow-up)](https://github.com/GioCC/MobiFlight-FirmwareSource/commit/c7a513768490a0f39ec852982837daff50c02284)

Ups, I wasn't aware that you already changed the InputShifter Code. But good to see that we did nearly the same modifications...
[my follow up](https://github.com/MobiFlight/MobiFlight-FirmwareSource/compare/main...elral:MobiFlight-FirmwareSource:InputShifter_CodeHarmonization)

GioCC, 2022-10-04:
Yes, I basically copied how it was already in the multiplexer code, which I found had very little to add.

elral, 2022-10-12:
I reopen this issue because for the AnalogIn the start up value is not set to the actual value.

GioCC, 2022-10-12:
@elral shouldn't this be more related to #202?

elral, 2022-10-12:
Hmhm, #202 deals with the retrigger function, this issue with avoiding serial messages on initial startup by initialising the device with it's status on startup (more explanations above). So I think it belongs to this issue.

I wrote:
> For analog in this is fixed by PR https://github.com/MobiFlight/MobiFlight-FirmwareSource/pull/199 where the average buffer is filled on startup.

but there is some more missing. Within the constructor `_lastValue` is set to `0` and in the first loop a change is detected as the averageBuffer is set to the actual analogIn value. So a serial message gets generated.
I added a comment (https://github.com/MobiFlight/MobiFlight-FirmwareSource/pull/199/files/0d7011e2d9f6ed7ef63fc5b55f74289484b45686#diff-9eb4e6794b83f2c6f1fca4a7e62355c388fc8aa3c4cb52067700a6527b4f7931) to the already merged PR #199 which might be not the best way. I guess you were not informed about this.

GioCC, 2022-10-12:
@elral Oh, I see it now - I was mistaken by focusing on the retrigger (which would set `_lastValue` correctly), but I missed the fact that the update loop starts _before_ a retrigger is requested.
But as I understand it, the problem should be that a (correct but) unwanted message is generated, not that the initial value sent is different from the actual value, is that correct?

elral, 2022-10-13:
> is that correct?

Yes.

> In hindsight, I guess you were probably describing the solution, not the issue.

First thread describes the problem. I see an issue also the right place to discuss (possible) solution(s).

PR #201 (2022-10-03): Improve getting correct device infos for connector from Mega
## Description of changes
Buttons and input shifters are initialized with the actual status of inputs.
This avoids serial messages an startup due to changed status of inputs in case they are closed (ON).

Fixes #200
PR #203 (2022-10-08): Use same retrigger logic for Multiplexer and InputShifter
Changes:

- [x] Added value initialization for digital input multiplexers
- [x] Introduce enum DO_TRIGGER and DONT_TRIGGER for both device types
- [x] Introduce poll method for both device types to reduce duplicate code
- [x] use same trigger method logic 

## Description of changes
relates to #201 
fixes #200
PR #212 (2022-10-13): Fix faulty initialization of analog value at startup
## Description of changes

Fixes #200 (reopened) but belongs to the scope of the changes made in PR #199.

- [X] Added initialization of actual value returned, as opposed as buffer values only, in the constructor.
closed 2022-10-13 by DocMoebiuz