MobiFlight/MobiFlight-FirmwareSource #202: Retrigger function does not initialize current value of analog potentiometers #898

neilenns, 2022-10-03:
Firmware copy of https://github.com/MobiFlight/MobiFlight-Connector/issues/898

**Describe the bug**
The retrigger function does not initialize the current value of analog potentiometers position.

**Steps To Reproduce**
Steps to reproduce the behavior:

Start Mobiflight with a potentiometer config assigned to a light dimmer in the sim and pot in mid position.
Press the retrigger button
Note lights in the sim did not sync to the pot position.
Expected behavior
Retrigger should synchronize the pot position in the sim.

**Actual behavior**
Potentiometer is ignored in the retrigger. Nothing happens.

**Details**

MobiFlight Version: 9.4.0.4
FlightSim: irrelevant
AddOn: irrelevant

**Additional context**
Workaround is to move the pot, which defeats the purpose of the retrigger function.

closed 2022-10-03 by DocMoebiuz

tigert, 2022-11-07:
I am not sure if this actually works right currently. I set a value via pot, quit mobiflight, and move the pot to a different position, and restart mobiflight - and the value stays in the old setting until I move the potentiometer (unless it has a high sensitivity setting and it jitters on its own which naturally triggers an event but is not going to help for stable pots or hall sensors)

GioCC, 2022-11-08:
Thanks for the heads-up, I'll have a look as soon as I'm able.

GioCC, 2022-11-08:
I just had a quick look at the Connector's code; as I recall, I would expect a Retrigger command to be issued as soon as the Connector is set to run (or possibly when a configuration is loaded), but I can't seem to find it. The only point where this happens, if I'm not mistaken, is when a `RetriggerInputAction` is created following a configuration setting in an XML file.
Can anybody with a better familiarity with the Connector code maybe have a look at this?

neilenns, 2022-11-08:
Agreed @GioCC from my look at the Connector code as well

GioCC, 2022-11-08:
Thanks @neilenns, although this is one of those cases I'd have preferred to be proven wrong... 😅
I think I'll have to follow more closely what happens during startup as soon as I can have a board setup before me.

PR #199 (2022-10-03): (#898)(Connector) Fix missing retrigger for ANALOG inputs at startup
## Description of changes
Fixes #202 
Fixes #898 (listed under Connector issues)

Included Analog input polling in the retrigger command.
An initial scan has been added to setup() to fill the value buffer for analog inputs before it is used
commit 40b56940 (2022-10-03): (#898)(Connector) Fix missing retrigger for ANALOG inputs at startup (#199)

fixes #202