MobiFlight/MobiFlight-FirmwareSource #204: Cleanup of multiplexer device code

GioCC, 2022-10-04:
Remove code that handles "lazy" mode from multiplexer to simplify and gain space.
That mode is not in use and neither was it going to be used anytime soon (or ever).
(PR is ready and will be pushed as soon as previous pending modifications are closed).

DocMoebiuz, 2022-10-09:
> Remove code that handles "lazy" mode from multiplexer to simplify and gain space. That mode is not in use and neither was it going to be used anytime soon (or ever). 

This statement indicates, that this change is not necessary. Please close this issue if there are no functional or non-functional reasons, @GioCC. Thanks!

GioCC, 2022-10-09:
Having totally useless code lying around and wasting space is no concern?

DocMoebiuz, 2022-10-09:
It might have been already late when i read the description. Sorry. I must have read it totally wrong.

Code that is not used and will not be used any time soon can be removed of course to save memory space and reduce code size.

@GioCC go for it.

GioCC, 2022-10-10:
No problem, that just seemed a little odd. I'll file the PR.
Thanks

PR #210 (2022-10-13): Removed "Lazy" mode from Multiplexer code (#204)
## Description of changes
Fixes #204
- Removed `Lazy` code branch and `mode` argument from `MFDigInMux::poll()`
- Removed setLazyMode() (and related enum) from `MFDigInMux.*`
- Removed `mode` argument from `DigInMux::Add`
closed 2022-10-13 by DocMoebiuz