MobiFlight/MobiFlight-FirmwareSource #205: Move device names storage from RAM buffer to EEPROM

GioCC, 2022-10-04:
Device names are currently copied to a RAM buffer when the configuration is loaded, and later retrieved from there when they have to be inserted in messages.
They could be retrieved directly from the EEPROM when required, allowing for 1500 bytes of RAM to be freed up with no penalty except a negligible increase of processing time.

_(This issue could later become irrelevant if/when names will be removed from message content, but since that's still under scrutiny, until then it makes sense to benefit from not wasting memory.)_

DocMoebiuz, 2022-10-09:
we decided we won't do this, but change the way how the messages are structured. I will create a new issue for that.

DocMoebiuz, 2022-10-09:
@elral had already created the issue: https://github.com/MobiFlight/MobiFlight-FirmwareSource/issues/207

closed 2022-10-09 by DocMoebiuz