MobiFlight/MobiFlight-FirmwareSource #207: Eliminate the nameBuffer for freeing up additional RAM

elral, 2022-10-06:
As discussed on Discord already the nameBuffer could be eliminated and instead of the name from an input device an "identifier" would be send. The connector have to know about the identifier and the relation to the input device name, therefore the user will not seen any differences. The benefit is that the nameBuffer stored in the RAM is not required anymore and additional RAM is available. Furthermore the serial traffic is reduced as the names could be "long" and instead just an identifier is transferred.

The identifier could be the x.th device from the same type which gets initialised. This information is already available from number of intialised decices, e.g. for the buttons `buttonsRegistered` in `buttons[buttonsRegistered] = new (allocateMemory(sizeof(MFButton))) MFButton(pin, name);`. Instead of saving the name `buttonsRegistered` would be saved and on an event transferred.
During processing the config there is always the same order of initialising input devices so there is also always the same `indentifier`.
Reading the config from the connector happens also in the same order, therefore the connector has a fix relation between the `identifier` (x.te device) and the name which is transferred during reading the config.
The same applies to all other input devices.

For backwards capability the connector has to check the firmware version and depending on this the connector has to expect names or identifiers.

GioCC, 2022-10-06:
I repeat here the comment I made on Discord about the choice of the ID.
Unless there's some point I missed, using the Arduino pin number (data pin or first pin for multi-pin devices) as ID rather than the config order index would be more convenient. 
In the firmware, the reference pin is readily available, while the config order index should be stored just for this purpose, wasting memory.
Besides being as unambiguous as the index, it is also absolute: imagine reading the messages for troubleshooting and trying to figure out which "Button at position 15 in the internal array (or in the configuration sequence)" is, vs. "Button attached to pin #28".

GioCC, 2022-10-06:
For outputs, the config order is more conveniently used as ID because it allows immediate addressing; otherwise, an additional lookup table (or a search) would be required in the firmware.
If it is paramount to keep the same criterion for inputs and outputs (which is admittedly desirable, although it has been peacefully different until now), it could be debated that in a typical installations there are probably more input devices than output, therefore it might make sense to change the outputs instead.

closed 2025-02-24 by elral