MobiFlight/MobiFlight-FirmwareSource #225: Support the Raspberry Pi Pico

neilenns, 2022-12-08:


neilenns, 2022-12-08:
Let's get that PR open 🥳

elral, 2022-12-13:
I have a branch for supporting the Raspberry Pi Pico which is already communicated on Discord.
Additionally there are some projects which uses an own PCB using the RP2040 processor. Compared to the original Pico these PCB's support more I/O's. This would mean that an additional .json file is required and also an additional MFBoards.h file with another `MOBIFLIGHT_TYPE`. Everything else is the same.
We could also think about to have one .json file with all available I/O's (and one MFBoards.h file) for both cases (and for boards which also uses the RP2040 providing more I/O's). But this would mean if the "original" Pico is used, the user could assign pins which are in fact not available. At least when the user wants to connect his devices he will recognize it, so nothing really is going wrong.

@DocMoebiuz @neilenns @GioCC @tigert what is your opinion?

GioCC, 2022-12-13:
Personally, I'd be fine with the single .json for both types; however, I would expect quite a few questions from inexperienced users puzzled by the fact that they can't find the "missing" GPIOs, therefore I'd be inclined to vote for different .jsons.

DocMoebiuz, 2022-12-13:
are the other Pico boards not custom boards? or comparable to Uno vs Nano? With this analogy I would prefer dedicated json files that describe the respective capabilities properly.

DocMoebiuz, 2022-12-13:
i am just having the idea to create a template repository on github for custom boards that comes with a prepared platform io environment and when you create a release it will generate a hex file and the appropriate json file which then can be easily dropped into mobiflight (as long as no special detection is required)

neilenns, 2022-12-13:
> I have a branch for supporting the Raspberry Pi Pico which is already communicated on Discor

Yes, that's why I opened this issue :) To get those changes merged into the main firmware.

> Additionally there are some projects which uses an own PCB using the RP2040 processor.

Do you mean that people can buy online from other vendors besides Raspberry Pi? Or do you mean boards that people designed themselves for MobiFlight (like I did)?

> But this would mean if the "original" Pico is used, the user could assign pins which are in fact not available. At least when the user wants to connect his devices he will recognize it, so nothing really is going wrong.

We should ship a file with the available pins that people are most likely to use, which is the Raspberry Pi Pico board from Raspberry Pi. If people want the additional pins from a different board they can just add them to the board.json file themselves, or as @DocMoebiuz we could have alternate versions in a github repo somewhere.

elral, 2022-12-14:
> Do you mean that people can buy online from other vendors besides Raspberry Pi? Or do you mean boards that people designed themselves for MobiFlight (like I did)?

Both, the custom boards with own PCB's but mainly all the boards in the meantime with a RP2040 on board.

> or comparable to Uno vs Nano?

Not completly. Uno vs. Nano have the "problem" that a different command set for AVRdude is required, and this was the reason to setup a new .json file and environment in PlatformIO. Here in this case all different boards are flashed the same way and also the number of devices will not differ. There is no reason (for most of the boards) to setup a new environment in PlatformIO. "Only" the number of I/O's differ between the boards where Uno vs. Nano had the same one.

But I am also fine to use different .json files for the different RP2040 boards.

neilenns, 2022-12-14:
Do all these clones report as the same VID/PID as the original pico? If so then having multiple board.json files shipped doesn't make a difference because we can't tell the boards apart.

elral, 2022-12-14:
That's a good question. I don't know this...

GioCC, 2022-12-14:
I have a few, I might have a look.

DocMoebiuz, 2022-12-14:
in this case the ambiguity mechanism should kick in and allow the user to select from the options

neilenns, 2022-12-14:
@GioCC Can you also look at the contents of the `INFO_UF2.TXT` file on the clones?

GioCC, 2022-12-15:
Ok, from [this AliEx seller](https://www.aliexpress.com/item/1005003796653297.html) I ordered these types:
Type-C 16MB Purple
Type-C Black Version    (with RGB LED)

both report `USB\VID_2E8A&PID_0003&REV_0100&MI_01`
and the `INFO_UF2.TXT` content is for both
```
UF2 Bootloader v3.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2
```

neilenns, 2022-12-15:
Not surprised at all. We can either:

1. Ship a single `.board.json` with a single firmware and decide whether that should have all pins or only the pins exposed on the true Raspberry Pi Pico.
2. Ship multiple `.board.json` files with multiple firmware builds that report a different `MobiFlight Type` and rely on the disambiguation experience to have the user pick what kind of board they have.

Personally I think it's simplest to just ship the single board with only the exposed pins on the true Raspberry Pi Pico. I can check and see if our JSON parser supports comments, and if it does we can include the extra pins commented out in the file for people that need them.

PR #228 (2023-01-05): Pico
This pull request adds the support for the Raspberry Pi Pico.
The supported board is the original board from the Raspberry Pi Foundation.

* A environment is added to the platformio.ini file
* A MFBoards.h file is added for the board specific definitions
* After writing to the eeprom these changes have to be commited, accordingly MFEEPROM.cpp/h is modified
* Pico requires it's own servo lib. Therefore the servo lib for Atmels is moved from the standard environment to the boards environment from the Atmel boards
* The Pico servo lib gets initialized with the same range as for the Atmels
*  [... cut]
closed 2023-01-05 by DocMoebiuz