MobiFlight/MobiFlight-FirmwareSource #235: Clean up the build scripts

neilenns, 2023-02-02:
While working on some build process stuff for one of my private firmware builds I discovered a few things that need cleaning up:

1. Need to explicitly define `python_version` when downloading python. It should be `3.9`.
2. The caching mechanism isn't right, see https://github.com/platformio/platformio-docs/blob/develop/integration/ci/github-actions.rst
3. No need to do `python -m pip install --upgrade pip` before doing the pip install.

Example of these changes in action is at https://github.com/neilenns/Collins-FMS-3000-Firmware/tree/main/.github/workflows

elral, 2023-11-08:
@neilenns Is this Issue still open?

neilenns, 2023-11-08:
Looks like it. I checked the build scripts and everyone of those issues is still present.

elral, 2023-11-08:
@neilenns could you do the pull request? I am not really familiar with that.

PR #281 (2023-11-18): Clean up the build scripts
Fixes #235

## Description of changes

* Explicitly set the python version to use
* Add proper caching for PlatformIO

Build time without cache: 7 minutes 13 seconds
Build time with cache: 3 minutes 34 seconds
commit e5466891 (2023-11-18): Clean up the build scripts (#281)

* Clean up the build scripts
Fixes #235

* Trigger rebuild

* Trying again with different key

* Try caching the toolchain folder

* Force rebuild

* Do the release build as well
closed 2023-11-18 by DocMoebiuz