MobiFlight/MobiFlight-FirmwareSource #259: Unused varaibles in LedControl.cpp

neilenns, 2023-09-20:
```
.pio/libdeps/mega/LedControl/src/LedControl.cpp: In member function 'void LedControl::clearDisplay(int)':
.pio/libdeps/mega/LedControl/src/LedControl.cpp:100:9: warning: variable 'offset' set but not used [-Wunused-but-set-variable]
     int offset;
         ^~~~~~
.pio/libdeps/mega/LedControl/src/LedControl.cpp: In member function 'void LedControl::setDigit(int, int, byte, boolean)':
.pio/libdeps/mega/LedControl/src/LedControl.cpp:163:9: warning: variable 'offset' set but not used [-Wunused-but-set-variable]
     int offset;
         ^~~~~~
.pio/libdeps/mega/LedControl/src/LedControl.cpp: In member function 'void LedControl::setChar(int, int, char, boolean)':
.pio/libdeps/mega/LedControl/src/LedControl.cpp:181:9: warning: variable 'offset' set but not used [-Wunused-but-set-variable]
     int offset;
         ^~~~~~
```

Either use them or remove them

elral, 2023-09-20:
There is already a pull request since ~9 month to avoid this warning ;)

neilenns, 2023-09-20:
Which one? I didn't see an open issue for this 😬

elral, 2023-09-20:
It`s a library, PR is in the repo for the LEDControl (not on my PC)

neilenns, 2023-09-20:
Ahhhh

neilenns, 2023-09-20:
Can't do this without the library changing, closing.

closed 2023-09-20 by neilenns