MobiFlight/MobiFlight-FirmwareSource #273: Buttons, encoders, and diginmux actions do not prevent power save mode

neilenns, 2023-11-10:
The button, encoder, and digmuxin code are missing calls to `setLastCommandMillis()`, which means interacting with any of those devices isn't sufficient to prevent power save mode from kicking in.

We should probably call `setLastCommandMillis()` for those input devices too.

Reported by @elral in discord.

neilenns, 2023-11-10:
This might be a good time to rethink this whole method. Instead of having it be based on things with the device, why isn't it just the desktop sending a keep-alive message as long as "run" mode is active? We already have a defined event for that and it isn't used for anything. Just send it to the connected modules every once and a while.

tigert, 2023-11-10:
I agree. It is not correct to assume all panels will get data sent to them frequently enough.

My example was an APU EGT gauge created with a stepper motor and led output pin for gauge lighting.

As APU is used mostly just on the ground, the gauge will sit static during most of the flights, with the backlight on but nothing changing as far it is concerned.

I think MF connector should send keepalive events to all panels, whether they actually receive data or not.

elral, 2023-11-10:
I would change it in the firmware to enable and disable the power saving mode and let the connector decide when and which panel gets one of these commands.
Maybe we can add Neil's proposal as a interim solution as long as the connector is not modified. But it wouldn't solve Tuomas problem...

neilenns, 2023-11-10:
> I would change it in the firmware to enable and disable the power saving mode and let the connector decide when and which panel gets one of these commands.

The change I opened basically kind of does this. Modules put themselves to sleep after 15 minutes of not receiving a stay awake message from the desktop.

PR #274 (2023-11-18): Support keep alive messages from the desktop app
Fixes #273

This PR fundamentally changes how the firmware knows when to keep displays awake.

Instead of relying on the firmware to track when various events happen (and missing some events, like buttons, encoders, and multiplexers), it simply relies on the desktop app to tell it to stay awake.

The companion desktop app change sends a keep alive message every 5 minutes. If it's been 15 minutes since one of those messages came in the firmware will put displays to sleep.

Bonus: Removing all the calls to `setLastCommandMillis` saves about 1k of  [... cut]
commit 4d1a944a (2023-11-18): Support keep alive messages from the desktop app (#274)

* Buttons, encoders, and diginmux actions do not prevent power save mode
Fixes #273

* Remove unneeded include

* Fix build break

* Clean up debug log messages

* Clean up comments and debug string

* Get rid of accidental change

* Add support for enabling power save mode
closed 2023-11-18 by DocMoebiuz