MobiFlight/MobiFlight-FirmwareSource #278: PR builds can't be used to test modern features in MobiFlight

neilenns, 2023-11-14:
PR builds have a version number that starts with `0.0`, which makes the desktop app think they are too old to be used with modern features like renaming, serial number regeneration, and custom devices. This makes testing PR firmware builds a pain.

Ideally we'd find a way to fix this that still preserves the "this is a PR build" look of the PR build version numbers. Maybe we should use `99.99.prbuild` as the number instead?

neilenns, 2023-11-14:
Another option would be to change the desktop app to treat all firmware versions below 1.0.0 as "supports everything". I'm honestly not sure which approach is better.

elral, 2023-11-14:
> I'm honestly not sure which approach is better.

Hmhm, me too.

neilenns, 2023-11-14:
> Hmhm, me too.

The only advantage I can come up with to doing it in the connector is there will be somewhere with explicit code and a comment that documents the behaviour.

Having said that, using `99.99.prbuild` in the build script could also have a comment that says it's set to that so it always passes the test.

🤷‍♂️

elral, 2023-11-15:
I don't know how big the effort for changing the connector will be. As there are already checks for the FW version it would be a straight forward solution to do it at this place. Setting the FW version `99.99.prbuild` would be on a seond place. But in this case I would change the FW during build process to `0.pr.prbult` so it can directly been seen to which PR this FW belongs to. And checking in the connector to `0`.

But if too much effort has to be spend, lets go with `99.pr.prbuilt`.

neilenns, 2023-11-15:
Connector change is easy. I agree, I'll go do the change over there and close this.

I like the idea of including the pr number in the build, but it'll be `0.pr.prbuild` to make it easier to identify the build as a test build on the connector side. I'll open a new issue to make that change.

closed 2023-11-15 by neilenns