MobiFlight/MobiFlight-FirmwareSource #284: Pin 1 gets set as output or input when a button/analogIn is defined

elral, 2023-11-22:
With PR #253 the fixed device numbers were removed. All devices from one type are now behind each other in the deviceBuffer to reserve sufficient memory and only the pointer for all devices is saved.

To get the pointer it is called: `outputs = new (allocateMemory(sizeof(MFOutput) * count)) MFOutput;`

So the constructor gets called and executed. As the definition of the constructor is `MFOutput(uint8_t pin = 1);`  and the code is:

```
MFOutput::MFOutput(uint8_t pin)
{
    _pin   = pin;
    _value = false;
    pinMode(_pin, OUTPUT); // set pin to input
    set(_value);
}
```

Pin 1 gets set as Output if calculating the required size in the devicebuffer.

The same applies for the AnlogIn.

neilenns, 2023-11-22:
Woah, subtle bug!

elral, 2023-11-22:
I reduced everything to the minimum:

```
#include <Arduino.h>
#include "Output.h"

void setup() {
  Serial.begin(115200);
  while(!Serial) delay(10);
  delay(1000);
  Serial.println("Serial initialized");
  Output::setupArray(4);
  Output::Add(2);
  Output::Add(3);
  Output::Add(18);
  Output::Add(19);
}

void loop() {
  Serial.println("Setting output to 255");
  Output::OnSet(2,255);
  Output::OnSet(3,255);
  delay(1000);
  Serial.println("Setting output to 0");
  Output::OnSet(2,0);
  Output::OnSet(3,0);
  delay(1000);
}
```

with slight adaptions to get it run without the CMDMessenger and the onSet() function.

And it still happens.
The good things are, I have a base to to debug it better or to ask earlephilhower, and it is NOT related to our code.

PR #286 (2023-11-23): init devices with attach() function and not via constructor
## Description of changes

For the devices `AnalogIn` and `Buttons` an `attach()` function ia added where all required init paramaters are transferred. These init parameters were part of the constructor.
Now calling the constructor will only set that the device is not initialized.

The calling of the constructor is since PR #253 also done to calculate the required memory for all devices of one type and to reserve it.  As no init parameters are available at this point the standard paramaters were taken and e.g. pin 1 for the output and [... cut]
closed 2023-11-23 by DocMoebiuz